flip555/bms_connector #6: Seplos 3.0 BMS

flip555, 2023-09-11:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Currently only for Seplos 2.0 Ascii Modbus

### Describe the solution you'd like

Seplos 3.0 Ascii Modbus option

### Describe alternatives you've considered

- 

### Additional context

[SEPLOS-3.0RS485BMS.Modbus.RTU.Protocol.1._20230822104940 (1).pdf](https://github.com/flip555/seplos_bms_ha/files/12575714/SEPLOS-3.0RS485BMS.Modbus.RTU.Protocol.1._20230822104940.1.pdf)

flip555, 2023-09-12:
Hello everyone,

I've developed a Python script based on the V3.0 Protocol Manual to decode data from a SEPLOS V3.0 BMS using the Modbus RTU protocol. Unfortunately, I don't have access to a V3.0 BMS for testing.

If any of you have a V3.0 BMS or can access one, I'd greatly appreciate it if you could give this code a run. You may need to adjust the COM port and pack address in the script according to your setup.

Please post any output, feedback, or issues you encounter in this Issues ticket. Your contributions will help validate and improve this decoder for the community.

Thanks!

```python
# SEPLOS V3.0 BMS Modbus RTU Protocol Decoder
# This script is designed to decode data from a SEPLOS V3.0 BMS using the Modbus RTU protocol.
# It establishes a serial connection to the BMS, sends Modbus requests, and decodes the responses.
# The decoded data points are defined in the MODBUS_REQUESTS dictionary.

# GitHub Repository: https://github.com/flip555/seplos_bms_ha
# Discussions and issues can be found on the GitHub discussion board.
import serial

# Set the COM port and pack address here
COM_PORT = '/dev/ttyUSB0'
PACK_ADDRESS = 0x01

# Modbus RTU requests configuration
MODBUS_REQUESTS = {
    0x1000: {"name": "Pack Voltage", "type": "UINT16", "scale": 0.01, "unit": "V"},
    0x1001: {"name": "Current", "type": "INT16", "scale": 0.01, "unit": "A"},
    0x1002: {"name": "Remaining capacity", "type": "UINT16", "scale": 0.1, "unit": "Ah"},
    0x1003: {"name": "Total Capacity", "type": "UINT16", "scale": 0.1, "unit": "Ah"},
    0x1004: {"name": "Total Discharge Capacity", "type": "UINT16", "scale": 1, "unit": "Ah"},
    0x1005: {"name": "SOC", "type": "UINT16", "scale": 0.1, "unit": "%"},
    0x1006: {"name": "SOH", "type": "UINT16", "scale": 0.1, "unit": "%"},
    0x1007: {"name": "Cycle", "type": "UINT16", "scale": 1, "unit": ""},
    0x1008: {"name": "Average of Cell Voltage", "type": "UINT16", "scale": 0.001, "unit": "V"},
    0x1009: {"name": "Average of Cell Temperature", "type": "UINT16", "scale": 0.1, "unit": "K"},
    0x100A: {"name": "Max Cell Voltage", "type": "UINT16", "scale": 0.001, "unit": "V"},
    0x100B: {"name": "Min Cell Voltage", "type": "UINT16", "scale": 0.001, "unit": "V"},
    0x100C: {"name": "Max Cell Temperature", "type": "UINT16", "scale": 0.1, "unit": "K"},
    0x100D: {"name": "Min Cell Temperature", "type": "UINT16", "scale": 0.1, "unit": "K"},
    0x100E: {"name": "reserve", "type": "", "scale": 1, "unit": ""},
    0x100F: {"name": "MaxDisCurt", "type": "UINT16", "scale": 1, "unit": "A"},
    0x1010: {"name": "MaxChgCurt", "type": "UINT16", "scale": 1, "unit": "A"},
    0x1100: {"name": "Cell1 Voltage", "type": "UINT16", "scale": 0.001, "unit": "V"},
    0x1101: {"name": "Cell2 Voltage", "type": "UINT16", "scale": 0.001, "unit": "V"},
    0x1102: {"name": "Cell3 Voltage", "type": "UINT16", "scale": 0.001, "unit": "V"},
    0x1103: {"name": "Cell4 Voltage", "type": "UINT16", "scale": 0.001, "unit": "V"},
    0x1104: {"name": "Cell5 Voltage", "type": "UINT16", "scale": 0.001, "unit": "V"},
    0x1105: {"name": "Cell6 Voltage", "type": "UINT16", "scale": 0.001, "unit": "V"},
    0x1106: {"name": "Cell7 Voltage", "type": "UINT16", "scale": 0.001, "unit": "V"},
    0x1107: {"name": "Cell8 Voltage", "type": "UINT16", "scale": 0.001, "unit": "V"},
    0x1108: {"name": "Cell9 Voltage", "type": "UINT16", "scale": 0.001, "unit": "V"},
    0x1109: {"name": "Cell10 Voltage", "type": "UINT16", "scale": 0.001, "unit": "V"},
    0x110A: {"name": "Cell11 Voltage", "type": "UINT16", "scale": 0.001, "unit": "V"},
    0x110B: {"name": "Cell12 Voltage", "type": "UINT16", "scale": 0.001, "unit": "V"},
    0x110C: {"name": "Cell13 Voltage", "type": "UINT16", "scale": 0.001, "unit": "V"},
    0x110D: {"name": "Cell14 Voltage", "type": "UINT16", "scale": 0.001, "unit": "V"},
    0x110E: {"name": "Cell15 Voltage", "type": "UINT16", "scale": 0.001, "unit": "V"},
    0x110F: {"name": "Cell16 Voltage", "type": "UINT16", "scale": 0.001, "unit": "V"},
    0x1110: {"name": "Cell temperature 1", "type": "UINT16", "scale": 0.1, "unit": "K"},
    0x1111: {"name": "Cell temperature 2", "type": "UINT16", "scale": 0.1, "unit": "K"},
    0x1112: {"name": "Cell temperature 3", "type": "UINT16", "scale": 0.1, "unit": "K"},
    0x1113: {"name": "Cell temperature 4", "type": "UINT16", "scale": 0.1, "unit": "K"},
    0x1118: {"name": "Environment Temperature", "type": "UINT16", "scale": 0.1, "unit": "K"},
    0x1119: {"name": "Power Temperature", "type": "UINT16", "scale": 0.1, "unit": "K"},
}

# Function to convert bytes to the appropriate data type based on the type field
def convert_bytes_to_data(data_type, byte1, byte2):
    if data_type == "UINT16":
        return (byte1 << 8) | byte2
    elif data_type == "INT16":
        value = (byte1 << 8) | byte2
        # Convert to signed integer
        if value & 0x8000:
            value -= 0x10000
        return value
    else:
        return None

# Serial communication setup
ser = serial.Serial(COM_PORT, baudrate=19200, parity=serial.PARITY_NONE, stopbits=1, bytesize=8, timeout=1)

try:
    # Iterate through the Modbus requests and send each one to the device
    for register, config in MODBUS_REQUESTS.items():
        # Build Modbus RTU request
        request = bytes([PACK_ADDRESS, 0x04, (register >> 8) & 0xFF, register & 0xFF, 0x00, 0x02, 0x00, 0x00])
        ser.write(request)
        response = ser.read(5)  # Assuming the response is 5 bytes

        # Extract and convert data from the response
        if len(response) == 5:
            data_value = convert_bytes_to_data(config["type"], response[3], response[4])
            if data_value is not None:
                scaled_value = data_value * config["scale"]
                print(f"{config['name']}: {scaled_value} {config['unit']}")

except serial.SerialException as e:
    print("Serial communication error:", str(e))
finally:
    ser.close()

Plumpop1, 2023-09-12:
Hi Flip. Email me at brno6@seznam.cz and I will give you remote access to the HA, where it is connected via the USB port Seplos BMS 3.0

flip555, 2023-09-12:
Hey there,

If you're up for it, you can try the following steps in your `configuration.yaml` to run the script temporarily.

```
shell_command:
  # Step 1: Install wget if not already installed
  install_wget: apk add wget

  # Step 2: Download the SEPLOS Python script
  download_seplos_script: wget -O SEPLOS_V3.0_BMS_Modbus_RTU_Protocol_Decoder.py https://raw.githubusercontent.com/flip555/seplos_bms_ha/f282b1432e25d20108f91c12c413cfd9bc38b831/assets/misc/SEPLOS_V3.0_BMS_Modbus_RTU_Protocol_Decoder.py

  # Step 3: Run the SEPLOS Python script
  run_seplos_script: python SEPLOS_V3.0_BMS_Modbus_RTU_Protocol_Decoder.py

  # Delete the downloaded SEPLOS Python script
  delete_seplos_script: rm -f SEPLOS_V3.0_BMS_Modbus_RTU_Protocol_Decoder.py
```

Once you've added these to your configuration, restart Home Assistant (HA), and then head to the 'Developer' section in HA. Under 'Services,' you can run each step one by one. If possible, please send me any output or logs generated during Step 3.

To clean up you can just run the delete_seplos_script command and remove the shell commands from configuration.

Feel free to give it a shot when you have the time. If you have any questions or run into any issues, just let me know.

Cheers!

Plumpop1, 2023-09-12:
➜  ~ python SEPLOS_V3.0_BMS_Modbus_RTU_Protocol_Decoder.py                                                                                                                                                             
Traceback (most recent call last):
  File "/root/SEPLOS_V3.0_BMS_Modbus_RTU_Protocol_Decoder.py", line 8, in <module>
    import serial
ModuleNotFoundError: No module named 'serial'
➜  ~

Plumpop1, 2023-09-12:
2023-09-12 23:02:15.237 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration seplos_bms_ha which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-09-12 23:02:15.239 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-09-12 23:02:37.061 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up seplos_bms_ha platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 359, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/seplos_bms_ha/sensor.py", line 168, in async_setup_entry
    telemetry, alarms, battery_address, battery_address_2 = await async_update_data()
                                                            ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/seplos_bms_ha/sensor.py", line 160, in async_update_data
    telemetry, _, battery_address_1 = extract_data_from_message(telemetry_data_str, True, False, True)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: not enough values to unpack (expected 3, got 2)
2023-09-12 23:02:37.068 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up seplos_bms_ha platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 359, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/seplos_bms_ha/sensor.py", line 168, in async_setup_entry
    telemetry, alarms, battery_address, battery_address_2 = await async_update_data()
                                                            ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/seplos_bms_ha/sensor.py", line 160, in async_update_data
    telemetry, _, battery_address_1 = extract_data_from_message(telemetry_data_str, True, False, True)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: not enough values to unpack (expected 3, got 2)
2023-09-12 23:03:41.822 WARNING (MainThread) [homeassistant.components.mqtt.mixins] MQTT entity name starts with the device name in your config {'unique_id': 'abc__switch_discharge', 'name': 'abc discharge', 'device_class': <SwitchDeviceClass.OUTLET: 'outlet'>, 'state_topic': 'abc/switch/discharge', 'device': {'identifiers': ['abc'], 'name': 'abc', 'connections': []}, 'command_topic': 'homeassistant/switch/abc/discharge/set', 'payload_on': 'ON', 'availability_mode': 'latest', 'retain': False, 'payload_not_available': 'offline', 'payload_available': 'online', 'qos': 0, 'encoding': 'utf-8', 'optimistic': False, 'payload_off': 'OFF', 'enabled_by_default': True}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'discharge'
2023-09-12 23:03:41.833 WARNING (MainThread) [homeassistant.components.mqtt.mixins] MQTT entity name starts with the device name in your config {'unique_id': 'abc__switch_discharge', 'name': 'abc discharge switch', 'device_class': <BinarySensorDeviceClass.POWER: 'power'>, 'expire_after': 20, 'device': {'identifiers': ['abc'], 'name': 'abc', 'connections': []}, 'state_topic': 'abc/switch/discharge', 'payload_on': 'ON', 'availability_mode': 'latest', 'payload_not_available': 'offline', 'payload_available': 'online', 'force_update': False, 'qos': 0, 'encoding': 'utf-8', 'payload_off': 'OFF', 'enabled_by_default': True}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'discharge switch'
2023-09-12 23:03:41.837 WARNING (MainThread) [homeassistant.components.mqtt.mixins] MQTT entity name starts with the device name in your config {'unique_id': 'abc__switch_charge', 'name': 'abc charge', 'device_class': <SwitchDeviceClass.OUTLET: 'outlet'>, 'state_topic': 'abc/switch/charge', 'device': {'identifiers': ['abc'], 'name': 'abc', 'connections': []}, 'command_topic': 'homeassistant/switch/abc/charge/set', 'payload_on': 'ON', 'availability_mode': 'latest', 'retain': False, 'payload_not_available': 'offline', 'payload_available': 'online', 'qos': 0, 'encoding': 'utf-8', 'optimistic': False, 'payload_off': 'OFF', 'enabled_by_default': True}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'charge'
2023-09-12 23:03:41.861 WARNING (MainThread) [homeassistant.components.mqtt.mixins] MQTT entity name starts with the device name in your config {'unique_id': 'abc__switch_charge', 'name': 'abc charge switch', 'device_class': <BinarySensorDeviceClass.POWER: 'power'>, 'expire_after': 20, 'device': {'identifiers': ['abc'], 'name': 'abc', 'connections': []}, 'state_topic': 'abc/switch/charge', 'payload_on': 'ON', 'availability_mode': 'latest', 'payload_not_available': 'offline', 'payload_available': 'online', 'force_update': False, 'qos': 0, 'encoding': 'utf-8', 'payload_off': 'OFF', 'enabled_by_default': True}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'charge switch'

flip555, 2023-09-12:
Can you add this config.yaml restart HA, install pyserial and try to run it again

`
install_pyserial: pip install pyserial
`

Do you have a V2 connected for the addon because thats probably why the erroring wrong data but not much error handling yet and doesn't process V3 at all yet.

I ran the V3 script on my V2 via the method above and it obviously has totally different data 

```
`stdout: |-
  Sending request for Pack Voltage (Register 0x1000)
  Request sent: 0104100000020000
  Response received: 
  Sending request for Current (Register 0x1001)
  Request sent: 0104100100020000
  Response received: 
  Sending request for Remaining capacity (Register 0x1002)
  Request sent: 0104100200020000
  Response received: 
  Sending request for Total Capacity (Register 0x1003)
  Request sent: 0104100300020000
  Response received: 
  Sending request for Total Discharge Capacity (Register 0x1004)
  Request sent: 0104100400020000
  Response received: 
  Sending request for SOC (Register 0x1005)
  Request sent: 0104100500020000
  Response received: 
  Sending request for SOH (Register 0x1006)
  Request sent: 0104100600020000
  Response received: 
  Sending request for Cycle (Register 0x1007)
  Request sent: 0104100700020000
  Response received: 
  Sending request for Average of Cell Voltage (Register 0x1008)
  Request sent: 0104100800020000
  Response received: 
  Sending request for Average of Cell Temperature (Register 0x1009)
  Request sent: 0104100900020000
  Response received: 
  Sending request for Max Cell Voltage (Register 0x100a)
  Request sent: 0104100a00020000
  Response received: 
  Sending request for Min Cell Voltage (Register 0x100b)
  Request sent: 0104100b00020000
  Response received: 
  Sending request for Max Cell Temperature (Register 0x100c)
  Request sent: 0104100c00020000
  Response received: 
  Sending request for Min Cell Temperature (Register 0x100d)
  Request sent: 0104100d00020000
  Response received: 
  Sending request for reserve (Register 0x100e)
  Request sent: 0104100e00020000
  Response received: 
  Sending request for MaxDisCurt (Register 0x100f)
  Request sent: 0104100f00020000
  Response received: 
  Sending request for MaxChgCurt (Register 0x1010)
  Request sent: 0104101000020000
  Response received: 
  Sending request for Cell1 Voltage (Register 0x1100)
  Request sent: 0104110000020000
  Response received: 
  Sending request for Cell2 Voltage (Register 0x1101)
  Request sent: 0104110100020000
  Response received: 
  Sending request for Cell3 Voltage (Register 0x1102)
  Request sent: 0104110200020000
  Response received: 
  Sending request for Cell4 Voltage (Register 0x1103)
  Request sent: 0104110300020000
  Response received: 
  Sending request for Cell5 Voltage (Register 0x1104)
  Request sent: 0104110400020000
  Response received: 
  Sending request for Cell6 Voltage (Register 0x1105)
  Request sent: 0104110500020000
  Response received: 
  Sending request for Cell7 Voltage (Register 0x1106)
  Request sent: 0104110600020000
  Response received: 
  Sending request for Cell8 Voltage (Register 0x1107)
  Request sent: 0104110700020000
  Response received: 
  Sending request for Cell9 Voltage (Register 0x1108)
  Request sent: 0104110800020000
  Response received: 
  Sending request for Cell10 Voltage (Register 0x1109)
  Request sent: 0104110900020000
  Response received: 
  Sending request for Cell11 Voltage (Register 0x110a)
  Request sent: 0104110a00020000
  Response received: 
  Sending request for Cell12 Voltage (Register 0x110b)
  Request sent: 0104110b00020000
  Response received: 
  Sending request for Cell13 Voltage (Register 0x110c)
  Request sent: 0104110c00020000
  Response received: 
  Sending request for Cell14 Voltage (Register 0x110d)
  Request sent: 0104110d00020000
  Response received: 
  Sending request for Cell15 Voltage (Register 0x110e)
  Request sent: 0104110e00020000
  Response received: 
  Sending request for Cell16 Voltage (Register 0x110f)
  Request sent: 0104110f00020000
  Response received: 
  Sending request for Cell temperature 1 (Register 0x1110)
  Request sent: 0104111000020000
  Response received: 
  Sending request for Cell temperature 2 (Register 0x1111)
  Request sent: 0104111100020000
  Response received: 
  Sending request for Cell temperature 3 (Register 0x1112)
  Request sent: 0104111200020000
  Response received: 
  Sending request for Cell temperature 4 (Register 0x1113)
  Request sent: 0104111300020000
  Response received: 
  Sending request for Environment Temperature (Register 0x1118)
  Request sent: 0104111800020000
  Response received: 
  Sending request for Power Temperature (Register 0x1119)
  Request sent: 0104111900020000
  Response received:
stderr: ""
returncode: 0`
```

Plumpop1, 2023-09-13:
➜  ~ python SEPLOS_V3.0_BMS_Modbus_RTU_Protocol_Decoder.py
Sending request for Pack Voltage (Register 0x1000)
Request sent: 0104100000020000
Response received:
Sending request for Current (Register 0x1001)
Request sent: 0104100100020000
Response received:
Sending request for Remaining capacity (Register 0x1002)
Request sent: 0104100200020000
Response received:
Sending request for Total Capacity (Register 0x1003)
Request sent: 0104100300020000
Response received:
Sending request for Total Discharge Capacity (Register 0x1004)
Request sent: 0104100400020000
Response received:
Sending request for SOC (Register 0x1005)
Request sent: 0104100500020000
Response received:
Sending request for SOH (Register 0x1006)
Request sent: 0104100600020000
Response received:
Sending request for Cycle (Register 0x1007)
Request sent: 0104100700020000
Response received:
Sending request for Average of Cell Voltage (Register 0x1008)
Request sent: 0104100800020000
Response received:
Sending request for Average of Cell Temperature (Register 0x1009)
Request sent: 0104100900020000
Response received:
Sending request for Max Cell Voltage (Register 0x100a)
Request sent: 0104100a00020000
Response received:
Sending request for Min Cell Voltage (Register 0x100b)
Request sent: 0104100b00020000
Response received:
Sending request for Max Cell Temperature (Register 0x100c)
Request sent: 0104100c00020000
Response received:
Sending request for Min Cell Temperature (Register 0x100d)
Request sent: 0104100d00020000
Response received:
Sending request for reserve (Register 0x100e)
Request sent: 0104100e00020000
Response received:
Sending request for MaxDisCurt (Register 0x100f)
Request sent: 0104100f00020000
Response received:
Sending request for MaxChgCurt (Register 0x1010)
Request sent: 0104101000020000
Response received:
Sending request for Cell1 Voltage (Register 0x1100)
Request sent: 0104110000020000
Response received:
Sending request for Cell2 Voltage (Register 0x1101)
Request sent: 0104110100020000
Response received:
Sending request for Cell3 Voltage (Register 0x1102)
Request sent: 0104110200020000
Response received:
Sending request for Cell4 Voltage (Register 0x1103)
Request sent: 0104110300020000
Response received:
Sending request for Cell5 Voltage (Register 0x1104)
Request sent: 0104110400020000
Response received:
Sending request for Cell6 Voltage (Register 0x1105)
Request sent: 0104110500020000
Response received:
Sending request for Cell7 Voltage (Register 0x1106)
Request sent: 0104110600020000
Response received:
Sending request for Cell8 Voltage (Register 0x1107)
Request sent: 0104110700020000
Response received:

Plumpop1, 2023-09-13:
Sending request for Cell9 Voltage (Register 0x1108)
Request sent: 0104110800020000
Response received:
Sending request for Cell10 Voltage (Register 0x1109)
Request sent: 0104110900020000
Response received:
Sending request for Cell11 Voltage (Register 0x110a)
Request sent: 0104110a00020000
Response received:
Sending request for Cell12 Voltage (Register 0x110b)
Request sent: 0104110b00020000
Response received:
Sending request for Cell13 Voltage (Register 0x110c)
Request sent: 0104110c00020000
Response received:
Sending request for Cell14 Voltage (Register 0x110d)
Request sent: 0104110d00020000
Response received:
Sending request for Cell15 Voltage (Register 0x110e)
Request sent: 0104110e00020000
Response received:
Sending request for Cell16 Voltage (Register 0x110f)
Request sent: 0104110f00020000
Response received:
Sending request for Cell temperature 1 (Register 0x1110)
Request sent: 0104111000020000
Response received:
Sending request for Cell temperature 2 (Register 0x1111)
Request sent: 0104111100020000
Response received:
Sending request for Cell temperature 3 (Register 0x1112)
Request sent: 0104111200020000
Response received:
Sending request for Cell temperature 4 (Register 0x1113)
Request sent: 0104111300020000
Response received:
Sending request for Environment Temperature (Register 0x1118)
Request sent: 0104111800020000
Response received:
Sending request for Power Temperature (Register 0x1119)
Request sent: 0104111900020000
Response received:
➜  ~
➜  ~

flip555, 2023-09-13:
Aww ok thanks
I've just sent you an email

flip555, 2023-09-13:
Thanks @Plumpop1 !

Just for reference here is where I got upto updated code is [https://github.com/flip555/seplos_bms_ha/tree/next-branch/assets/misc/SEPLOS_V3.0_BMS_Modbus_RTU_Protocol_Decoder.py](https://github.com/flip555/seplos_bms_ha/tree/next-branch/assets/misc/SEPLOS_V3.0_BMS_Modbus_RTU_Protocol_Decoder.py)

Current output:
```
~ # python config/SEPLOS_V3.0_BMS_Modbus_RTU_Protocol_Decoder.py
Pack Address: 0x0
Request sent: 0004100000127516
Response received: 000424149afedd389a3a98005d03c603e700070ce00b930ce50cdb0b940b9300000096009603e8c05f
Pack Voltage (V): 52.74
Current (A): -2.91
Remaining Capacity (Ah): 144.9
Total Capacity (Ah): 150.0
Total Discharge Capacity (Ah): 9.3
SOC (%): 96.60000000000001
SOH (%): 99.9
Cycle: 7
Avg Cell Voltage (V): 3.2960000000000003
Avg Cell Temperature (°C): 29.63
Max Cell Voltage (V): 3.301
Min Cell Voltage (V): 3.291
Max Cell Temperature (°C): 29.64
Min Cell Temperature (°C): 29.63
Request sent (PIB): 00041100001274ea
Response received (PIB): 0004240cde0ce50ce00cde0ce40cdd0cdb0ce10ce20ce20ce40cdf0ce20cdf0cde0ce20b940b934f08
PIB Data:
Cell1 Voltage (V): 3.294
Cell2 Voltage (V): 3.301
Cell3 Voltage (V): 3.2960000000000003
Cell4 Voltage (V): 3.294
Cell5 Voltage (V): 3.3000000000000003
Cell6 Voltage (V): 3.293
Cell7 Voltage (V): 3.291
Cell8 Voltage (V): 3.297
Cell9 Voltage (V): 3.298
Cell10 Voltage (V): 3.298
Cell11 Voltage (V): 3.3000000000000003
Cell12 Voltage (V): 3.295
Cell13 Voltage (V): 3.298
Cell14 Voltage (V): 3.295
Cell15 Voltage (V): 3.294
Cell16 Voltage (V): 3.298
Request sent (PIC): 00041200001274ae
Response received (PIC): 0084029301
Warning: Not enough data in PIC response. The response may be incomplete.

```
PIB is missing Cell Temps, Environment Temp and Power Temp
PIC didn't look like it worked at all just probably an error string

Full Code: 
```
import serial
import serial.tools.list_ports

# Set the COM port and pack addresses here
COM_PORT = '/dev/ttyUSB0'
PACK_ADDRESSES = [0x00]  # Addresses for battery packs 0 and 1

# Check if the specified COM port is available
available_ports = [port.device for port in serial.tools.list_ports.comports()]

if COM_PORT not in available_ports:
    print(f"Error: {COM_PORT} is not an available serial port.")
    exit(1)

# Function to convert bytes to the appropriate data type based on the type field
def convert_bytes_to_data(data_type, byte1, byte2):
    if data_type == "UINT16":
[... 228 lines cut ...]
                        "Current Event Code 1": int(data_fields_pic[12]) & 0xFF,
                    }

                    # Print PIC data
                    print("PIC Data:")
                    for field, value in pic_data.items():
                        print(f"{field}: {value}")
                else:
                    print("Warning: Not enough data in PIC response. The response may be incomplete.")

            else:
                print("Error: CRC check failed for PIC")
        else:
            print("Error: Unexpected response format for PIC")

        # Continue with any further processing or data collection you need to do

        # ... (rest of the code)

except serial.SerialException as e:
    print("Serial communication error:", str(e))
finally:
    ser.close()

```

closed 2024-02-19 by flip555