flip555/bms_connector #9: Urgent Fixes

flip555, 2023-09-11:
### BMS Version 🔄

v2

### Connection Method 🌐

USB-RS485

### Pre-submission Checklist ✅

- [x] This issue is focused on a single problem (For multiple issues, create separate reports).
- [x] I've checked for [existing issues](https://github.com/flip555/seplos_bms_ha/issues?q=is%3Aissue+label%3A%22Bug%22) and confirmed mine is unique.

### What's happening? 🤔

- fix sensor data type - getting graphs on all the alarms instead of string
- change sensor name values! - WARNING all sensor names will be changing!
- incorporate a few more sensors 

### Steps to Reproduce 🔍

1.
2.
3.
...

Plumpop1, 2023-09-11:
I have problem. 
[image]

flip555, 2023-09-12:
> I have problem. [image]

Looks like you just need to type the path of your USB-RS485 in. 

I have updated the code with a few fixes so it should populate it with /dev/ttyUSB0 but you may need to change the 0 at the end depending on your system.

closed 2023-09-12 by flip555