flip555/bms_connector #14: Add to HACS store

flip555, 2023-09-12:
### BMS Version 🔄

-

### Connection Method 🌐

USB-RS485

### Problem Description 🤔

-

### Proposed Solution 🛠️

-

Plumpop1, 2023-10-11:
Hi 

Logger: custom_components.bms_connector.sensor
Source: custom_components/bms_connector/sensor.py:53
Integration: BMS Connector (documentation, issues)
First occurred: 9. října 2023 v 23:54:06 (3 occurrences)
Last logged: 13:42:27

Error setting up BMS sensors: not enough values to unpack (expected 5, got 2)

flip555, 2023-10-11:
This could be one of a few things.

USB COM Port - if it doesn't see the correct USB port or doesn't have permissions to the USB port (permissions would be more Docker HA than HAOS though)

Battery address, I think we only support 0x00 for V2 and 0x01 for V3 at the moment. 

If this was already working but stopped its more likely USB port delete the entry in Settings > Integrations > BMS Connector and readd it with new USB port

Plumpop1, 2023-10-11:
It has never been connected before and is a V2 10C model