flip555/bms_connector #30: https://github.com/flip555/seplos_bms_ha is not a valid repository

ornolio, 2023-09-25:
### BMS Version 🔄

seplos 2

### Connection Method 🌐

USB-RS485

### Pre-submission Checklist ✅

- [X] This issue is focused on a single problem (For multiple issues, create separate reports).
- [X] I've checked for [existing issues](https://github.com/flip555/seplos_bms_ha/issues?q=is%3Aissue+label%3A%22Bug%22) and confirmed mine is unique.

### What's happening? 🤔

when try to add Hacs app I get:

https://github.com/flip555/seplos_bms_ha is not a valid add-on repository. 
HA CORE 2023.9.2
Please help.

### Steps to Reproduce 🔍

1.
2.
3.
...

flip555, 2023-09-25:
Please try to use the new repository instead

https://github.com/flip555/bms_connector

ornolio, 2023-09-25:
Same error using:

https://github.com/flip555/bms_connector

ornolio, 2023-09-25:
same error using:
https://github.com/flip555/bms_connector

closed 2023-09-25 by ornolio