flip555/bms_connector #31: bug #30 is not closed.

ornolio, 2023-09-27:
### BMS Version 🔄

seplos v 2

### Connection Method 🌐

USB-RS485

### Pre-submission Checklist ✅

- [X] This issue is focused on a single problem (For multiple issues, create separate reports).
- [X] I've checked for [existing issues](https://github.com/flip555/seplos_bms_ha/issues?q=is%3Aissue+label%3A%22Bug%22) and confirmed mine is unique.

### What's happening? 🤔

Your suggestion to try:

https://github.com/flip555/bms_connector

doesn't solve the problem: I got the same error. This is not a valud.

### Steps to Reproduce 🔍

1.
2.
3.
...

flip555, 2023-09-27:
I have been unable to replicate this issue. Could you please provide logs or screenshots for further investigation? Additionally, please make sure you are adding the custom repository through the HACS store and not via the Home Assistant addon store.

ornolio, 2023-09-27:
Really I use the Home Assistant add on store, not the HACS store. sorry, my fault. 
I'll learn how to use the HACS Store instead and I'll let you know.
Meanwhile let me close this "not-a-bug".
Thank you.

closed 2023-09-27 by ornolio

ornolio, 2023-09-28:
Ok, I succeed now with the installation inside HACS Store. I apologize again bothering you. Thank You for the help.