flip555/bms_connector #52: Parallel BMS

ornolio, 2024-01-31:
### BMS Version 🔄

seplos v.2

### Connection Method 🌐

USB-RS485

### Problem Description 🤔

Currently many installations use battery packs in parallel each wirh its own  BMS; in my case 4 packs with  4 x v.2. BMS RS485 ports of Seplos are linked by UTP cables in a daisy chain and Seplos switches accordingly with Seplos installation manual to addees each pack and set total number of packs in parallel. One end of RS485 cable goes to PC Window APP provided by Seplis thtrogh a RS485 to USB converter. 

The  RS485 connection is also to give the Master Seplos BMSf the capability to drive the load distribution among each pack in parallel.

The App provides either a view of all parameters of each  single selected pack by clicking the pack number on the Window screen or a recap of  all packs  parameters in anoher screen. 

Currently it appears BMS connector doesn't work in this configuration. 

Could you please put an eye on the protocol and see if it would be possible would  to have this upgrade c@BMS Connector v. 2? Thank you in advance

### Proposed Solution 🛠️

I suspect some the Seplos protocol adds some preamble string to identify the total number of packs in parallel and the pack address of following data.

ornolio, 2024-01-31:
Thank you so much for your reply.

Yes, I believe to be able to provide you tomorrow with a Log of multiple packs communication to the Windows App using its Log option.

Best Regards.
Pier

ornolio, 2024-02-01:
[DataRecord_20240201_034205.xls](https://github.com/flip555/bms_connector/files/14127946/DataRecord_20240201_034205.xls)

here you are. 
This log file has ben formatted by the Seplos BMS Windows App on a stack of three 1p16s Lifepo4 14kWh packs with Seplos BMS v. 2.
Packs are in daisy chain with UTP-RJ45 cbles connected to RS485 ports.
BMS 8bit Switches set are: Pack1 10000000, Pack2 0100000; Pack3 11000000 
Let me know you would like some USB Sniffer Log. In such a case do you have a windows app to get it ?
Best Regards

flip555, 2024-02-19:
Thanks for that!

I have released a version for testing https://github.com/flip555/bms_connector/releases/tag/v1.1.0-TESTING-ONLY

You should be able to update to this by going to HACS > BMS Connector > Redownload > Enable Beta Versions and selecting v1.1.0-TESTING-ONLY

[image]

There is quite a bit missing from it, and needs some streamlining but would be great if you could test it before I add in the rest of the data as I'm only able to test my single battery pack. It does seem to work using test data

[image]

ornolio, 2024-02-21:
Right now I'm far away from the battery stack to reconnect RS485 daisy chain and test new version. I'll do in a few days, next we.
Regards.
p.

flip555, 2024-03-01:
I'm pretty sure parallel won't work with current code from what I've seen on issue #60 - Looks like the main BMS polls the parallel BMS's automatically which all gets dumped into the same serial output (commands and responses) so I need change that part of the code to listen, poll, sort and verify the data before processing/submitting to HA, something I've been playing with today.

That document is very nice, looks like it has complete data tables too some of which I was missing. Bluetooth was something I was looking at so I guess I finally have a starting point for that too.

I do remember 42H is actually 62H in other documents I think it was the pylontech BMS's they're all very similar.

ornolio, 2024-03-02:
For sure the Master BMS does poll the slave BMSs and gets data on its own Master RS485 port; then if I link it through a single RS485-USB adaptor to a Single USB port of a windows PC running the Seplos's Battery Monitor V2.1.8, I can view on this app dashboard data of master and slave BMS (see pictures in the case of 1 Master + 1 Slave.
[image]
[image]
[image]

flip555, 2024-03-02:
Hmm interesting. I've been looking and testing remotely on a paralleled system. It has 3 packs but even listening to the RS485 port just shows pack 1 & 2 - Don't seem to be able to send commands to it though as its all scrolling far too quick. 

Your .xls communication log posted before also only shows packs 1, 2 and 3 but was a 4 pack battery. 

Those screenshots show Pack 1 & 2 from a 2 pack so I'm even more confused now as its not missing Pack 0 😂 

Looks like there was/is a problem with paralleled batteries but watching Andy @ Off Grid Garage it was working like your 2pack and was switching from Pack 0 to Pack 1, then he enabled dip 5 for CAN and it stopped. 

Seems to be dip switches and firmware/hardware versions 

Could you confirm DIP switch setup on your setup?

Andy seemed to have 
```
1P: 00000000
2P: 10000000
```

and it worked in Battery Monitor he then changed it to enable CAN for his victron system to 
```
1P: 00001000
2P: 10000000
```

and it stopped working in battery monitor, I'm just thinking a parallel pack starts with 0x01 when CAN is enabled 

I'm thinking you have:
```
1P: 10001000
2P: 01000000
```

Or maybe you have automatic dip switch setting enabled?

ornolio, 2024-03-02:
Seplos dip switches are very confusing. Depending if you use CAN or not dip switches change, requiring to set dip 5,6,7,8 of first pack to total no. of packs in parallel; in such a case dips 1,2,3,4 starts from 1 instead of 0.

I do not use CAN to communicate with inverter.
Previous Windows monitor screens refers  2 packs, with following dip switches setup::

'00000000'. pack1
'10000000'  pack2

tomorrow afternoon I'll be at packs location, so I'll  configure the whole 4 packs to get  the  4 packs log of windows monitor and test the bms connector behaviour at the same dip switch configuration . as well.

ornolio, 2024-03-03:
Attached I'm sending you the log of the Seplos Battery Monitor, using a single RS485-USB and 4 batteries in parallel. Dip configuration are:

'10000000' pack1
'01000000' pack2
'11000000' pack3
'00100000' pack4

No CAN used.
[DataRecord_20240303_033049.xls](https://github.com/flip555/bms_connector/files/14473305/DataRecord_20240303_033049.xls)

ornolio, 2024-03-03:
Attached are also the screen shots of the Seplos bms monitor. Dip switches config is the same
[image]
[image]
[image]
[image]
[image]

flip555, 2024-03-04:
Thanks, I think it'll be best to rework it a bit so it adds whichever modbus parts it receives to HA and skips (not errors and exits) when missing a full set of data (42H, 44H, 47H, 51H), then it can poll later if missing/has old data. 

In the "BMS Modbus Responses" sensor you should have Attributes tab, this should have the latest modbus responses, for each battery pack so I guess I'm nearly where I need to be.

ornolio, 2024-03-13:
tested with two packs in parallel
dip switch at:
'10000000'. pack1
'01000000' pack2
RS485 on usb0
set number of pack 2 when adding hub.
set name prefix to 'Pack1'
I get 164 entities. of second pack.

Setting number of pack 1 when adding hub
set prefix to 'Packk2'
I get 165 entities (one more) always of the second pack, same of before.

entity names  result to contain 0x00 in both cases.

ornolio, 2024-04-10:
[image]
update to v. 1.1.9 test only
having 3 battery packs with seplos version 2
RS485 linked in daisy chain
dip seitches to:
10000000
01000000
11000000
00100000
RS485 if first pack to USB adapter to Raspi4 usb port 0
git the following:
[image]
[image]
[image]

ornolio, 2024-04-10:
last 3 pictures on reverse order.