flip555/bms_connector #57: entities not available with 4 HUBs for 4 Seplos v. 2

ornolio, 2024-02-18:
### BMS Version 🔄

1.05 

### Connection Method 🌐

USB-RS485

### Pre-submission Checklist ✅

- [X] This issue is focused on a single problem (For multiple issues, create separate reports).
- [X] I've checked for [existing issues](https://github.com/flip555/seplos_bms_ha/issues?q=is%3Aissue+label%3A%22Bug%22) and confirmed mine is unique.

### What's happening? 🤔

I have a stack of 4x14kWh LifePO4 batteries in parallel, controlled by 4xBMS Seplos v. 2, using 4 HUBs of the BMS Connector running on a Raspberry Pi4 with Homeassistant through the ports USB0, USB1, USB2, USB3:
I configured 4 HUBs (Pack1, Pack2, Pack3 and Pack4) on BMS connector; each HUB acquires 193 entities from its own Seplos.

So for each HUB in the home assistant dashboard I show:
-Current,
-Voltage,
-SOC%
then for each HUB I add and display the totals of:
-Residual Capacity,
-Rated Capacity
-(Rated Capacity -Residual Capacity)
In total I use 20 entities and show 15 of which 3 are calculated.
I disabled all the other unused entities of the 4 HUBs.
So of 193x4=772 entities acquired by the 4 HUBs, I only use 20 of them, the others are disabled.

I) Unfortunately the dashboard is unable to show any of these 20 entities, which are: "not available".

II)Naturally even with all 772 entities enabled, the dashboard does not show these 20 entities which are all "not available".

III)If in the BMS Connector configuration I disable at least 2 HUBs (it doesn't matter which ones), then the dashboard shows the correct values of the two HUBs not disabled.

lV)If I disable polling of at least 2 HUBs in the BMS Connector configuration, then the dashboard shows the "static" values of the entities of the 2 disabled HUBs and the "dynamic" values of the remaining enabled entities.

I imagine there is a sampling time problem in the case of a high number of entities transmitted by each Seplos BMS.

Do you have any chance to fix this problem with 4 HUBs configured?
Thanks for your attention.

### Steps to Reproduce 🔍

1.configure 4 HUB USB-RS485 with 4 Seplos v. 2 (not inteconnected)
2.
3.
...

flip555, 2024-02-19:
Sounds like this is probably a naming conflict within the code, different USB entries fighting over the same entity id's as I don't think the USB port was included in the names. I have started to add the USB device to the unique IDs in the new release

ornolio, 2024-02-21:
entity names contain packs number (Pack1, Pack2...), so they should not overlap. (see picture).

Right now I'm not close to battery stack. I'll do it in a few days.

Best Regards.
p.

ornolio, 2024-02-21:
[image]
[image]
[image]

ornolio, 2024-02-21:
I installed the test version: https://github.com/flip555/bms_connector/releases/tag/v1.1.0-TESTING-ONLY
it works with 3 HUBs, the 4 one turns  all the entities of the 4 HUBS into "unavailable".status.

flip555, 2024-02-24:
Thanks. The test version only has about 20 sensors added per hub/pack at the moment the rest would be Unavailable as they are not being provided.

Looking at what you have I'm pretty sure you'd have to delete/disable and re-add the entries but all ID's have changed in the code at the moment.

Trying to work out the best method to name them so they don't overlap, I think it'll be best to reimplement the custom prefix again like you have Pack1_ Pack2_ as its much tidier than trying to autogenerate everything as is in the test version.

I'll spend some time changing it back to user defined prefixes and release a new test version.

ornolio, 2024-02-24:
Thank You Sir,

meanwhile I still continue investigation on my problem. 
I would like to perform some more tests concerning  Raspi4 USB hardware power drain  when all 4 USB ports have RS485 to USB adaptors plugged in, although  PI power supply provides 4A.

Awaiting for any news.
Best Regards.

ornolio, 2024-02-29:
Problem is  due to low power of RASPI USB ports when 4 adaptors RS485 to USB  are all plugged in. 

Even if RASPI PI4 has been powered with a 4Amps power supply.

I solved this issue using a USB HUB HAT  to Raspi self powered with an additional.4Amps ps.

BMS works as a charm.

closed 2024-02-29 by ornolio