flip555/bms_connector #60: Bug

Plumpop1, 2024-02-26:
### BMS Version 🔄

V 2.0

### Connection Method 🌐

USB-RS485

### Pre-submission Checklist ✅

- [X] This issue is focused on a single problem (For multiple issues, create separate reports).
- [X] I've checked for [existing issues](https://github.com/flip555/seplos_bms_ha/issues?q=is%3Aissue+label%3A%22Bug%22) and confirmed mine is unique.

### What's happening? 🤔

Entities are not loaded

### Steps to Reproduce 🔍

1.
2.
3.
...

### Debug logs 📜

```text
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 344, in _async_setup_platform
await asyncio.shield(task)
File "/config/custom_components/bms_connector/sensor.py", line 22, in async_setup_entry
await hass.data[DOMAIN]["HOME_ENERGY_HUB_SENSOR_COORDINATOR"+entry_id].async_refresh()
~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyError: 'HOME_ENERGY_HUB_SENSOR_COORDINATORdf7cf633016891488a69c9e081246d5c'
2024-02-26 20:27:19.777 ERROR (MainThread) [homeassistant.components.binary_sensor] Error while setting up bms_connector platform for binary_sensor
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 344, in _async_setup_platform
await asyncio.shield(task)
File "/config/custom_components/bms_connector/binary_sensor.py", line 22, in async_setup_entry
await hass.data[DOMAIN]["HOME_ENERGY_HUB_SENSOR_COORDINATOR"+entry_id].async_refresh()
~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyError: 'HOME_ENERGY_HUB_SENSOR_COORDINATORdf7cf633016891488a69c9e081246d5c'
2024-02-26 20:27:24.079 DEBUG (MainThread) [custom_components.bms_connector] Seplos V2 BMS Device Selected..
2024-02-26 20:27:24.080 DEBUG (MainThread) [custom_components.bms_connector.modules.bms.seplos.v2] ['~20004642E00200FD37\r', '~20004644E00200FD35\r', '~20004647E00200FD32\r', '~20004651E00200FD37\r']
2024-02-26 20:27:24.092 DEBUG (MainThread) [custom_components.bms_connector.modules.bms.seplos.v2] ~20004642E00200FD37
2024-02-26 20:27:24.603 DEBUG (MainThread) [custom_components.bms_connector.modules.bms.seplos.v2] ~20004644E00200FD35
2024-02-26 20:27:25.112 ERROR (MainThread) [custom_components.bms_connector] Error setting up BMS Connector: 'utf-8' codec can't decode byte 0xa8 in position 12: invalid start byte
2024-02-26 20:27:25.115 INFO (MainThread) [homeassistant.components.sensor] Setting up bms_connector.sensor
2024-02-26 20:27:25.116 INFO (MainThread) [homeassistant.components.binary_sensor] Setting up bms_connector.binary_sensor
2024-02-26 20:27:25.118 INFO (MainThread) [homeassistant.components.select] Setting up bms_connector.select
2024-02-26 20:27:25.119 INFO (MainThread) [homeassistant.components.number] Setting up bms_connector.number
2024-02-26 20:27:25.123 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up bms_connector platform for sensor
```

flip555, 2024-02-27:
Looks like you're using one of the test releases which should work fine with V2 but looks like its failing somewhere after sending the 44H CID2 command.

Could you update to the v1.1.4-TESTING-ONLY version and provide the debug log again as I've added some more debugging logs and error handling around that part of the code.

Thanks

Plumpop1, 2024-02-27:
2024-02-27 16:49:48.001 WARNING (MainThread) [homeassistant.components.number] custom_components.bms_connector.number::YourNumberClass is overriding deprecated methods on an instance of NumberEntity, this is not valid and will be unsupported from Home Assistant 2022.10. Please report it to the custom integration author

2024-02-27 17:25:17.647 ERROR (MainThread) [custom_components.bms_connector.modules.bms.seplos.v2] Error decoding response: 'utf-8' codec can't decode byte 0xcb in position 109: invalid continuation byte
2024-02-27 17:25:17.647 DEBUG (MainThread) [custom_components.bms_connector.modules.bms.seplos.v2] Sending Modbus Command: ~20014647E00201FD30
2024-02-27 17:25:18.153 DEBUG (MainThread) [custom_components.bms_connector.modules.bms.seplos.v2] Raw Response Received: b'~200146008152013C0DAC0D980B540C1C0E2E0D980AF00C1C0D4805DC15E01518122012C01680151810E012C0189C17D40C9F0C810ABF0ADD0CD10C9F0A470AAB0CB30C810A470AC90CD10C9F0A150AAB0AAB0B0F0D030CD10AAB0AC90D530D350A470AAB0E2F0DFD0E930DFD4E204C2CAFECB0B45208ADF88AD007D0753036B01B321E1E140A100A0A1E3C0505010A0A1EF0300F0562506409000D0008FFFFFF3FBF9FAF9F313130312D4C4E313220B27C\r~2001465A0000FD9D\r~2001465AC04000010CC80CC30BB60BA8F583146E4CAA7530028E146F00000000000003028041EFF9\r'
2024-02-27 17:25:18.154 DEBUG (MainThread) [custom_components.bms_connector.modules.bms.seplos.v2] Decoded Response Received: ~200146008152013C0DAC0D980B540C1C0E2E0D980AF00C1C0D4805DC15E01518122012C01680151810E012C0189C17D40C9F0C810ABF0ADD0CD10C9F0A470AAB0CB30C810A470AC90CD10C9F0A150AAB0AAB0B0F0D030CD10AAB0AC90D530D350A470AAB0E2F0DFD0E930DFD4E204C2CAFECB0B45208ADF88AD007D0753036B01B321E1E140A100A0A1E3C0505010A0A1EF0300F0562506409000D0008FFFFFF3FBF9FAF9F313130312D4C4E313220B27C~2001465A0000FD9D~2001465AC04000010CC80CC30BB60BA8F583146E4CAA7530028E146F00000000000003028041EFF9
2024-02-27 17:25:18.154 DEBUG (MainThread) [custom_components.bms_connector.modules.bms.seplos.v2] Sending Modbus Command: ~20014651E00201FD36
2024-02-27 17:25:18.658 DEBUG (MainThread) [custom_components.bms_connector.modules.bms.seplos.v2] Raw Response Received: b'~200146020000FDB1\r~2002465A0000FD9C\r~2002465AC04000010CC70CC30BB30BA9F53C146F4DA675300296146F00000000000003028041F007\r'
2024-02-27 17:25:18.659 DEBUG (MainThread) [custom_components.bms_connector.modules.bms.seplos.v2] Decoded Response Received: ~200146020000FDB1~2002465A0000FD9C~2002465AC04000010CC70CC30BB30BA9F53C146F4DA675300296146F00000000000003028041F007
2024-02-27 17:25:18.665 DEBUG (MainThread) [custom_components.bms_connector.modules.bms.seplos.v2] Complete Modbus Responses Received: ['~2001460010960001100CC80CC70CC60CC40CC40CC30CC30CC30CC30CC40CC30CC50CC50CC40CC70CC8060BAA0BA80BB60BAC0BD90BC1F586146E4CAB0A7530028E7530001103E81470FFE7000009080839DC4E~2001465A0000FD9D~2001465AC04000010CC80CC30BB60BA8F586146E4CAB7530028E147000000000000003028041F00A', '7e3230303134363030383036323030303131303030303030303030303030303030303030303030303030303030303030303030303630303030303030303030303030303030313430303030303030303030303030333030303030313030303030303030307e3230303234363558cbebdab3b3e7357e323030313436303530303030464441450d', '~200146008152013C0DAC0D980B540C1C0E2E0D980AF00C1C0D4805DC15E01518122012C01680151810E012C0189C17D40C9F0C810ABF0ADD0CD10C9F0A470AAB0CB30C810A470AC90CD10C9F0A150AAB0AAB0B0F0D030CD10AAB0AC90D530D350A470AAB0E2F0DFD0E930DFD4E204C2CAFECB0B45208ADF88AD007D0753036B01B321E1E140A100A0A1E3C0505010A0A1EF0300F0562506409000D0008FFFFFF3FBF9FAF9F313130312D4C4E313220B27C~2001465A0000FD9D~2001465AC04000010CC80CC30BB60BA8F583146E4CAA7530028E146F00000000000003028041EFF9', '~200146020000FDB1~2002465A0000FD9C~2002465AC04000010CC70CC30BB30BA9F53C146F4DA675300296146F00000000000003028041F007']
2024-02-27 17:25:18.667 ERROR (MainThread) [custom_components.bms_connector] Error setting up BMS Connector: non-hexadecimal number found in fromhex() arg at position 4
2024-02-27 17:25:18.670 INFO (MainThread) [homeassistant.components.sensor] Setting up bms_connector.sensor
2024-02-27 17:25:18.671 INFO (MainThread) [homeassistant.components.binary_sensor] Setting up bms_connector.binary_sensor
2024-02-27 17:25:18.672 INFO (MainThread) [homeassistant.components.select] Setting up bms_connector.select
2024-02-27 17:25:18.673 INFO (MainThread) [homeassistant.components.number] Setting up bms_connector.number
2024-02-27 17:25:18.675 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up bms_connector platform for sensor
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 344, in _async_setup_platform
await asyncio.shield(task)
File "/config/custom_components/bms_connector/sensor.py", line 22, in async_setup_entry
await hass.data[DOMAIN]["HOME_ENERGY_HUB_SENSOR_COORDINATOR"+entry_id].async_refresh()
~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyError: 'HOME_ENERGY_HUB_SENSOR_COORDINATORdf7cf633016891488a69c9e081246d5c'
2024-02-27 17:25:18.687 ERROR (MainThread) [homeassistant.components.binary_sensor] Error while setting up bms_connector platform for binary_sensor
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 344, in _async_setup_platform
await asyncio.shield(task)
File "/config/custom_components/bms_connector/binary_sensor.py", line 22, in async_setup_entry
await hass.data[DOMAIN]["HOME_ENERGY_HUB_SENSOR_COORDINATOR"+entry_id].async_refresh()
~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyError: 'HOME_ENERGY_HUB_SENSOR_COORDINATORdf7cf633016891488a69c9e081246d5c'

flip555, 2024-02-27:
The data returned isn't that same format as what my V2 returns - Is this a V2 BMS? could you load it up in Battery Monitor and download the communications log

Data returned by your BMS
```
'2001460010960001100CC80CC70CC60CC40CC40CC30CC30CC30CC30CC40CC30CC50CC50CC40CC70CC8060BAA0BA80BB60BAC0BD90BC1F586146E4CAB0A7530028E7530001103E81470FFE7000009080839DC4E2001465A0000FD9D2001465AC04000010CC80CC30BB60BA8F586146E4CAB7530028E147000000000000003028041F00A', 

'7e3230303134363030383036323030303131303030303030303030303030303030303030303030303030303030303030303030303630303030303030303030303030303030313430303030303030303030303030333030303030313030303030303030307e3230303234363558cbebdab3b3e7357e323030313436303530303030464441450d',

'200146008152013C0DAC0D980B540C1C0E2E0D980AF00C1C0D4805DC15E01518122012C01680151810E012C0189C17D40C9F0C810ABF0ADD0CD10C9F0A470AAB0CB30C810A470AC90CD10C9F0A150AAB0AAB0B0F0D030CD10AAB0AC90D530D350A470AAB0E2F0DFD0E930DFD4E204C2CAFECB0B45208ADF88AD007D0753036B01B321E1E140A100A0A1E3C0505010A0A1EF0300F0562506409000D0008FFFFFF3FBF9FAF9F313130312D4C4E313220B27C2001465A0000FD9D2001465AC04000010CC80CC30BB60BA8F583146E4CAA7530028E146F00000000000003028041EFF9',

'200146020000FDB12002465A0000FD9C~2002465AC04000010CC70CC30BB30BA9F53C146F4DA675300296146F00000000000003028041F007'
```

Data returned by my BMS
```
'~2000460010960000100CF00CF10CF10CF10CF00CF20CF00CF00CEF0CF00CF00CEF0CEF0CEF0CF10CF0060B6B0B710B770B800B3E0B21FBD914B34C5D0A5D98032F59D8008E03E814B5FFED00003C2D3A17DBD1',

'~20004600806200001000000000000000000000000000000000060000000000000000140000000000000300000100000000000000000001EB33', 

'~200046008152003C0DAC0D480B540BB80E420E100A8C0B540DE805DC16761644122012C016801676104011F816F816A80C9F0C810ABF0ADD0CD10C9F0A470AAB0CB30C810A470AC90CD10C9F0A150AAB0AAB0B0F0C9F0C810AAB0AC90D030CD10A470AAB0E2F0DFD0E930DFD4E204C2CAFECB0B45208ADF88AD007D059D859D81B321E1E140A100A0A1E3C0505010A0A1EF0300F0560506409000D0008FFFEFF3FBF819F1E313130312D5350373620B2A0',

'~20004600C040313130312D5350373620100443414E3A56696374726F6E202020202020202020F098'

```

Plumpop1, 2024-02-27:
2024-02-27 16:48:56.635 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration bms_connector which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-02-27 16:48:56.639 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration battery_notes which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-02-27 16:48:56.644 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration rpi_gpio which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-02-27 16:48:56.649 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-02-27 16:48:56.653 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration solarman which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-02-27 16:48:56.656 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration cz_energy_spot_prices which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-02-27 16:48:57.759 INFO (MainThread) [homeassistant.setup] Setup of domain logger took 0.1 seconds
2024-02-27 16:48:57.760 INFO (MainThread) [homeassistant.bootstrap] Setting up frontend: {'frontend'}
2024-02-27 16:48:57.913 INFO (MainThread) [homeassistant.setup] Setting up device_automation
2024-02-27 16:48:57.914 INFO (MainThread) [homeassistant.setup] Setup of domain device_automation took 0.0 seconds
2024-02-27 16:48:58.750 INFO (MainThread) [homeassistant.setup] Setting up lovelace
2024-02-27 16:48:58.765 INFO (MainThread) [homeassistant.setup] Setting up http
2024-02-27 16:48:58.794 INFO (MainThread) [homeassistant.setup] Setup of domain lovelace took 0.0 seconds
2024-02-27 16:48:58.948 INFO (MainThread) [homeassistant.setup] Setup of domain http took 0.2 seconds
2024-02-27 16:48:59.010 INFO (MainThread) [homeassistant.setup] Setting up api
2024-02-27 16:48:59.020 INFO (MainThread) [homeassistant.setup] Setup of domain api took 0.0 seconds
2024-02-27 16:48:59.025 INFO (MainThread) [homeassistant.setup] Setting up auth
2024-02-27 16:48:59.030 INFO (MainThread) [homeassistant.setup] Setup of domain auth took 0.0 seconds
2024-02-27 16:48:59.087 INFO (MainThread) [homeassistant.setup] Setting up config
2024-02-27 16:48:59.153 INFO (MainThread) [homeassistant.setup] Setting up diagnostics
2024-02-27 16:48:59.234 INFO (MainThread) [homeassistant.setup] Setting up file_upload
2024-02-27 16:48:59.235 INFO (MainThread) [homeassistant.setup] Setup of domain file_upload took 0.0 seconds
2024-02-27 16:48:59.313 INFO (MainThread) [homeassistant.setup] Setting up repairs
2024-02-27 16:48:59.316 INFO (MainThread) [homeassistant.setup] Setup of domain repairs took 0.0 seconds
2024-02-27 16:48:59.321 INFO (MainThread) [homeassistant.setup] Setting up websocket_api
2024-02-27 16:48:59.323 INFO (MainThread) [homeassistant.setup] Setup of domain websocket_api took 0.0 seconds
2024-02-27 16:48:59.682 INFO (MainThread) [homeassistant.setup] Setting up image_upload
2024-02-27 16:49:00.372 INFO (MainThread) [homeassistant.setup] Setup of domain config took 1.3 seconds
2024-02-27 16:49:00.377 INFO (MainThread) [homeassistant.setup] Setup of domain diagnostics took 1.2 seconds
2024-02-27 16:49:00.394 INFO (MainThread) [homeassistant.setup] Setup of domain image_upload took 0.7 seconds
2024-02-27 16:49:00.679 INFO (MainThread) [homeassistant.setup] Setting up search
2024-02-27 16:49:00.680 INFO (MainThread) [homeassistant.setup] Setup of domain search took 0.0 seconds
2024-02-27 16:49:06.122 INFO (MainThread) [homeassistant.setup] Setting up analytics
2024-02-27 16:49:06.130 INFO (MainThread) [homeassistant.setup] Setting up person
2024-02-27 16:49:06.137 INFO (MainThread) [homeassistant.setup] Setup of domain analytics took 0.0 seconds
2024-02-27 16:49:06.153 INFO (MainThread) [homeassistant.setup] Setup of domain person took 0.0 seconds
2024-02-27 16:49:06.161 INFO (MainThread) [homeassistant.setup] Setting up onboarding
2024-02-27 16:49:06.167 INFO (MainThread) [homeassistant.setup] Setup of domain onboarding took 0.0 seconds
2024-02-27 16:49:06.178 INFO (MainThread) [homeassistant.setup] Setting up frontend
2024-02-27 16:49:06.214 INFO (MainThread) [homeassistant.setup] Setup of domain frontend took 0.0 seconds
2024-02-27 16:49:06.222 INFO (MainThread) [homeassistant.bootstrap] Setting up recorder: {'recorder'}
2024-02-27 16:49:06.232 INFO (MainThread) [homeassistant.setup] Setting up recorder
2024-02-27 16:49:06.304 INFO (MainThread) [homeassistant.components.http] Now listening on port 8123
2024-02-27 16:49:06.502 INFO (MainThread) [homeassistant.setup] Setup of domain recorder took 0.3 seconds
2024-02-27 16:49:06.510 INFO (MainThread) [homeassistant.bootstrap] Setting up stage 1: {'ssdp', 'network', 'http', 'cloud', 'websocket_api', 'bluetooth', 'api', 'dhcp', 'usb', 'webhook', 'zeroconf', 'hassio'}
[... cut ...]
2024-02-27 22:49:31.950 INFO (MainThread) [homeassistant.components.websocket_api.http.connection] [546598994752] Initialized trigger
2024-02-27 22:49:34.870 INFO (MainThread) [homeassistant.components.automation.nizky_tarif] Initialized trigger Nízký Tarif
2024-02-27 22:49:39.857 INFO (MainThread) [homeassistant.components.automation.nizky_tarif] Nízký Tarif: Running automation actions
2024-02-27 22:49:39.858 INFO (MainThread) [homeassistant.components.automation.nizky_tarif] Nízký Tarif: Executing step device automation
2024-02-27 22:50:00.689 INFO (MainThread) [custom_components.hacs] Loading known repositories

Plumpop1, 2024-02-27:
Seplos V2.0 10E 3x

Plumpop1, 2024-02-28:
HACS (Home Assistant Community Store)

Version: 1.34.0
This is a custom integration
If you have any issues with this you need to open an issue here:
https://github.com/hacs/integration/issues
-------------------------------------------------------------------

2024-02-28 16:51:56.520 WARNING (MainThread) [homeassistant.components.binary_sensor] Setup of binary_sensor platform rpi_gpio is taking over 10 seconds.
2024-02-28 16:51:56.522 INFO (MainThread) [homeassistant.setup] Setup of domain system_health took 11.9 seconds
2024-02-28 16:51:56.567 INFO (MainThread) [homeassistant.setup] Setup of domain media_source took 8.2 seconds
2024-02-28 16:51:56.581 INFO (MainThread) [homeassistant.setup] Setting up select
2024-02-28 16:51:56.589 INFO (MainThread) [homeassistant.setup] Setup of domain select took 0.0 seconds
2024-02-28 16:51:56.595 INFO (MainThread) [homeassistant.setup] Setting up number
2024-02-28 16:51:56.597 INFO (MainThread) [homeassistant.setup] Setup of domain number took 0.0 seconds
2024-02-28 16:51:56.598 INFO (MainThread) [homeassistant.setup] Setup of domain logbook took 7.5 seconds
2024-02-28 16:51:56.678 INFO (MainThread) [homeassistant.setup] Setup of domain input_text took 16.4 seconds
2024-02-28 16:51:56.703 INFO (MainThread) [homeassistant.setup] Setup of domain input_button took 16.4 seconds
2024-02-28 16:51:56.732 INFO (MainThread) [homeassistant.setup] Setup of domain tag took 15.9 seconds
2024-02-28 16:51:56.732 DEBUG (MainThread) [custom_components.bms_connector.modules.bms.seplos.v2] Raw Response Received: 
2024-02-28 16:51:56.733 DEBUG (MainThread) [custom_components.bms_connector.modules.bms.seplos.v2] Decoded Response Received: 
2024-02-28 16:51:56.733 DEBUG (MainThread) [custom_components.bms_connector.modules.bms.seplos.v2] Sending Modbus Command: ~20014644E00201FD33
2024-02-28 16:51:56.755 DEBUG (MainThread) [custom_components.bms_connector.modules.bms.seplos.v2] Raw Response Received: ~2001460010960001100CD40CD00CCD0CCF0CCD0CCB0CD20CCF0CCE0CCF0CCF0CCF0CCF0CCF0CD20CD1060B910B910B980B940BC20B9D0000147F2EC60A7530018F7530001203E81482FFCFFF350916086FDBC5
2024-02-28 16:51:56.756 DEBUG (MainThread) [custom_components.bms_connector.modules.bms.seplos.v2] Decoded Response Received: ~2001460010960001100CD40CD00CCD0CCF0CCD0CCB0CD20CCF0CCE0CCF0CCF0CCF0CCF0CCF0CD20CD1060B910B910B980B940BC20B9D0000147F2EC60A7530018F7530001203E81482FFCFFF350916086FDBC5
2024-02-28 16:51:56.756 DEBUG (MainThread) [custom_components.bms_connector.modules.bms.seplos.v2] Sending Modbus Command: ~20014647E00201FD30
2024-02-28 16:51:56.763 DEBUG (MainThread) [custom_components.bms_connector.modules.bms.seplos.v2] Raw Response Received: ~2001465A0000FD9D
2024-02-28 16:51:56.763 DEBUG (MainThread) [custom_components.bms_connector.modules.bms.seplos.v2] Decoded Response Received: ~2001465A0000FD9D
2024-02-28 16:51:56.764 DEBUG (MainThread) [custom_components.bms_connector.modules.bms.seplos.v2] Sending Modbus Command: ~20014651E00201FD36
2024-02-28 16:51:56.777 DEBUG (MainThread) [custom_components.bms_connector.modules.bms.seplos.v2] Raw Response Received: ~2001465AC04000010CD40CCB0B980B91FFEB147F2EC67530018F148200000000000003028010EFF3
2024-02-28 16:51:56.778 DEBUG (MainThread) [custom_components.bms_connector.modules.bms.seplos.v2] Decoded Response Received: ~2001465AC04000010CD40CCB0B980B91FFEB147F2EC67530018F148200000000000003028010EFF3
2024-02-28 16:51:56.783 DEBUG (MainThread) [custom_components.bms_connector.modules.bms.seplos.v2] Complete Modbus Responses Received: ['', '~2001460010960001100CD40CD00CCD0CCF0CCD0CCB0CD20CCF0CCE0CCF0CCF0CCF0CCF0CCF0CD20CD1060B910B910B980B940BC20B9D0000147F2EC60A7530018F7530001203E81482FFCFFF350916086FDBC5', '~2001465A0000FD9D', '~2001465AC04000010CD40CCB0B980B91FFEB147F2EC67530018F148200000000000003028010EFF3']
2024-02-28 16:51:56.784 DEBUG (MainThread) [custom_components.bms_connector.modules.bms.seplos.v2] Sending Modbus Commands: ['~20024642E00202FD33\r', '~20024644E00202FD31\r', '~20024647E00202FD2E\r', '~20024651E00202FD35\r']
2024-02-28 16:51:56.797 DEBUG (MainThread) [custom_components.bms_connector.modules.bms.seplos.v2] Sending Modbus Command: ~20024642E00202FD33
2024-02-28 16:51:56.836 WARNING (MainThread) [homeassistant.components.number] custom_components.bms_connector.number::YourNumberClass is overriding deprecated methods on an instance of NumberEntity, this is not valid and will be unsupported from Home Assistant 2022.10. Please report it to the custom integration author
2024-02-28 16:51:59.570 INFO (MainThread) [homeassistant.setup] Setting up media_player
2024-02-28 16:51:59.635 INFO (MainThread) [homeassistant.setup] Setup of domain media_player took 0.1 seconds
2024-02-28 16:51:59.640 INFO (SyncWorker_2) [homeassistant.loader] Loaded todo from homeassistant.components.todo
2024-02-28 16:51:59.706 INFO (MainThread) [homeassistant.setup] Setup of domain input_boolean took 18.6 seconds
2024-02-28 16:51:59.727 INFO (MainThread) [homeassistant.setup] Setup of domain input_number took 18.6 seconds
2024-02-28 16:51:59.728 INFO (MainThread) [homeassistant.setup] Setup of domain zone took 19.1 seconds
2024-02-28 16:51:59.739 INFO (MainThread) [homeassistant.setup] Setup of domain timer took 18.6 seconds
2024-02-28 16:51:59.756 INFO (MainThread) [homeassistant.setup] Setup of domain application_credentials took 11.4 seconds
2024-02-28 16:51:59.873 DEBUG (MainThread) [custom_components.bms_connector.modules.bms.seplos.v2] Raw Response Received: 
2024-02-28 16:51:59.874 DEBUG (MainThread) [custom_components.bms_connector.modules.bms.seplos.v2] Decoded Response Received: 
2024-02-28 16:51:59.874 DEBUG (MainThread) [custom_components.bms_connector.modules.bms.seplos.v2] Sending Modbus Command: ~20024644E00202FD31
2024-02-28 16:51:59.880 ERROR (MainThread) [custom_components.bms_connector] Error setting up BMS Connector: 'utf-8' codec can't decode byte 0xb2 in position 0: invalid start byte

flip555, 2024-03-01:
Ok I still have quite a bit of work to do on the code but I think V1.1.7-TESTING-ONLY will get us closer, I've changed the way it handles and checks the modbus data