flip555/bms_connector #78: Seplos v3 multi Bms

jolly12f, 2024-06-08:
### BMS Version 🔄

V3

### Connection Method 🌐

USB-RS485

### Problem Description 🤔

values ​​are not recognized

### Proposed Solution 🛠️

Is it possible to view two battery packs connected to each other?

### Alternatives Considered 🔄

No

### Additional context 📄

No

ErikCoen, 2024-08-04:
I have now a 2 Seplos v3 in parallel mode. However this BMS connector app will not recognize it. What should be modified in the settings?

shavermcspud, 2024-10-22:
I also have the same issue with 2x Seplos BMS vs 3

creaturanelu, 2025-10-04:
I also have 3 packs in parallel , im getting wrong info, like 3000% soc then 300, the ln 4500% etc, all data are wrong with large numbers..

Antho27520, 2026-05-23:
I just downloaded and installed version V1.3.0 and it's the same thing.

milemanolov, 2026-10-28:
Setup

Home Assistant with the BMS Connector custom integration.

Three Seplos Mason-280 BMS units.

All three BMS configurations use the same USB-to-RS485 adapter (/dev/ttyUSB0).

Serial settings: 19200 baud, 8 data bits, no parity, 1 stop bit.

BMS addresses: 0x01, 0x02, and 0x03.

Observed behavior

When only one BMS configuration is enabled, communication works reliably.

When all three configurations are enabled, serial communication errors appear in the logs, including:

device reports readiness to read but returned no data (device disconnected or multiple access on port?)

Modbus CRC mismatches.

Corrupted or incomplete responses.

Increasing the polling interval of the third BMS from 10 to 30 seconds did not resolve the issue.

Suspected cause

Each BMS configuration appears to have its own polling coordinator, while all three configurations access the same serial port through the same transport module.

The serial transport opens the port for each request, but there was no shared lock to serialize access between the configurations. Concurrent requests may therefore interfere with each other, resulting in incomplete responses and CRC errors.

This is the most likely explanation based on the observed behavior, although it has not been conclusively proven.

Change tested

A shared threading.Lock was added to the serial transport module. The original send_serial_command() function was renamed to _send_serial_command_unlocked(), and a new send_serial_command() wrapper was added to acquire the shared lock before executing the original function.

The modified file passed Python syntax validation, and Home Assistant Core was restarted to load the change.

Result

After the change, the battery readings are stable, but some serial communication and CRC errors still appear in the logs.

The change appears to have improved stability, but it has not eliminated all communication errors.

Further investigation

It may be worth checking whether the lock should cover the entire request/response transaction, including any address-discovery operations, and whether the serial port should be managed by a single shared transport instance rather than opened separately for each request.

Duplicate sensor unique-ID warnings have also been observed and may need to be investigated separately.

Current status: The system is usable, and the stable battery readings are acceptable for now.
[https://imgur.com/a/qdDLQD4](url)